pyros-dev/pyros #6: Sync message_conversion module with other libraries

asmodehn, 2015-06-19:
Transferred from https://github.com/asmodehn/rostful/issues/14

jihoonl, 2015-06-19:
Coming from https://github.com/asmodehn/rostful/issues/14

rosbridge_library.internal.message_converter and ros_loader would be better since it has the most test cases and under my control.

It may provide too much functionality though.

asmodehn, 2017-01-31:
I wrote a new library to handle the parsing in a much more flexible way. It might be more complex but it uses a python parsing package already used in other places. check https://github.com/asmodehn/pyros-schemas

One requirement for this issue however is to be able to confirm that our message serialization is behaving the same as / equivalent to rospy/rosgen use.

A good way would be to use property-based testing (check https://github.com/HypothesisWorks/hypothesis-python) to test both library in the exact same way, for all possible values.