pyros-dev/pyros #17: Adding support for wildcard '*' for exposing topics / services

asmodehn, 2015-07-12:
wildcard (regex ? ) support for exposing topics / services would help in many cases.
- exposing all topics/services from one actions
- exposing all topics/services from a node
- exposing all topics/services from a rapp
- etc.

asmodehn, 2015-08-01:
I had a look at the commit 472f92d9e57906426ed03d5c7f803eeb3e647914, it seems to me that you're doing something unnecessary : 
if REGEX -> do this
if not REGEX -> do that

AFAIK "/long/topic/full/name" is also a regex, that will match only one topic : /long/topic/full/name
So it seems we can have regex in all cases and we don't need yet another "topics_match" on top of "topics_args" to store topics...

heuristicus, 2015-08-03:
Removed the separate lists in 63c6e6570da08b87430412362bc0cc31ada0d511.

asmodehn, 2015-12-08:
this is in and seems to work fine.

closed 2015-12-08 by asmodehn