pyros-dev/pyros #19: Subprocesses cannot publish to topics

heuristicus, 2015-07-28:


heuristicus, 2015-07-28:
According to [this comment](https://github.com/ros/ros_comm/blob/78aa1c4ec879668f686aba873f52968b1bd7aa73/clients/rospy/src/rospy/topics.py#L146), topics and services can be initialised before the node - does this affect anything?

heuristicus, 2015-07-28:
Information about added publishers or subscribers is sent to the master node [here](https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/impl/registration.py#L420).

heuristicus, 2015-07-28:
`subscriber_listener` [here](https://github.com/ros/ros_comm/blob/78aa1c4ec879668f686aba873f52968b1bd7aa73/clients/rospy/src/rospy/topics.py#L805) is `None` when initialising the chatter topic in subprocess.

heuristicus, 2015-07-28:
Might be able to use the `subscriber_listener` to trigger addition of the subscription topic to the connections list of the publisher?

heuristicus, 2015-07-29:
`tcpros_base` is where checks are made to determine subscribers and publishers (?). This is run in a separate thread, with the `run` function, which runs until the node is shutdown.

heuristicus, 2015-07-29:
`tcpros_pubsub::topic_connection_handler` is where connections for topics are created, both subscribers and publishers.

heuristicus, 2015-07-29:
`topics::TopicManager::acquire_impl` creates the subscriber/publisher.

Line 1149 `get_registration_listeners().notify_added(resolved_name, ps.type, reg_type)`  seems to be important, appears to call `reg_added` function of `registration::RegManager`, which informs the master about the new topic or subscriber. `publisher_update` is then called, which modifies the `updates` list.

heuristicus, 2015-08-03:
@asmodehn looks like there's nothing we can do about this. I suppose we can close this.

asmodehn, 2015-08-03:
@heuristicus yes lets keep fixing the current design then.

closed 2015-08-03 by asmodehn