pyros-dev/pyros #20: Clean up ros interface

heuristicus, 2015-07-31:
Currently `add_topic` and `expose_topic` have very similar functionality, and having both makes things very confusing.

The `topics_args` are used in a couple of places, but they don't seem to follow what is their expected use - if anything is added to those lists, it is not displayed in the REST interface.

asmodehn, 2015-07-31:
Hmmm we probably should refactor this part of the code... but before doing
that we should have all unit tests in place ( regarding parameters passing
/vs/ available topics/services ) to make sure we don't break anything in
the process.

asmodehn, 2015-12-08:
The interface has been changed as part of the refactoring to 0.1

closed 2015-12-08 by asmodehn