pyros-dev/pyros #52: Test Hanging on Jenkins

stonier, 2016-01-24:
Output from a bulid farm hang on this rospy test:

```
$ cat nosetests-src.pyros.rosinterface.rostests.testService.py.xml 
<?xml version="1.0" encoding="UTF-8"?><testsuite name="nosetests" tests="1" errors="1" failures="0" skip="0"><testcase classname="&lt;nose.suite.ContextSuite context=pyros.rosinterface.rostests" name="testService&gt;:setup" time="0.000"><error type="socket.error" message="[Errno 111] Connection refused&#10;-------------------- &gt;&gt; begin captured logging &lt;&lt; --------------------&#10;root: WARNING: ZMQ : Protobuf message implementation not found. Using pickle based protocol&#10;rospy.topics: INFO: topicmanager initialized&#10;rosout: WARNING: Missing rocon codebase. Rocon features disabled&#10;--------------------- &gt;&gt; end captured logging &lt;&lt; ---------------------"><![CDATA[  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 208, in run
    self.setUp()
  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 291, in setUp
    self.setupContext(ancestor)
  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 314, in setupContext
    try_run(context, names)
  File "/usr/lib/python2.7/dist-packages/nose/util.py", line 471, in try_run
    return func()
  File "/var/lib/jenkins/bootstrap/src/pyros/src/pyros/rosinterface/rostests/testService.py", line 38, in setup_module
    launch = roslaunch.scriptapi.ROSLaunch()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/scriptapi.py", line 61, in __init__
    uuid = master.getParam('/run_id')
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosgraph/masterapi.py", line 206, in getParam
    return self._succeed(self.handle.getParam(self.caller_id, key))
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
'[Errno 111] Connection refused\n-------------------- >> begin captured logging << --------------------\nroot: WARNING: ZMQ : Protobuf message implementation not found. Using pickle based protocol\nrospy.topics: INFO: topicmanager initialized\nrosout: WARNING: Missing rocon codebase. Rocon features disabled\n--------------------- >> end captured logging << ---------------------']]></error></testcase></testsuite>jenkins@chimek:~/bootstrap/build/test_results/pyros$ 
jenkins@chimek:~/bootstrap/build/test_results/pyros$
```

asmodehn, 2016-01-24:
Oh thanks for the report.
Problem likely related with this : https://github.com/ros/ros_comm/pull/711

Looks like the jenkins versiondoesnt have this patch but i thought i put a
workaround in pyros-setup for now...

I ll need to double check all versions on monday (rospy and pyros-setup on
my machine vs jenkins)

stonier, 2016-01-24:
Probably implicitly understood, but I forgot to mention that it is only happening sometimes...not every run.

asmodehn, 2016-01-29:
Although I dont know the root cause, this solved it : 
https://github.com/asmodehn/pyros/commit/c46f478478ee9b8859a0e091a35bd3a3f278f1bd

closed 2016-01-29 by asmodehn