pyros-dev/pyros #57: Odd import error

stonier, 2016-01-29:
The code here looks ok, but it triggers a failed import when running (`gocart_sim.launch`):

```
[2016-01-29 15:55:35,120: WARNING/MainProcess] Error: could not import RoconInterface - disabling. No module named roconinterface.rocon_interface
WARNING:root:Error: could not import RoconInterface - disabling. No module named roconinterface.rocon_interface
```

asmodehn, 2016-02-01:
This is an internal warning of pyros.
Since I haven't kept all the rocon things uptodate, the rocon part of pyros (rapp + interactions handling) fails and disables itself.

I have been thinking about dropping it. Gopher is not using it anyway.
And it could be done in another project, using rostful underneath.

asmodehn, 2016-02-05:
Planning to drop rocon support https://github.com/asmodehn/pyros/issues/61

closed 2016-02-05 by asmodehn