pyros-dev/pyros #100: Topic_extract in pyros_client does not working after rosinstance reinit.

dhirajdhule, 2016-10-17:
I am trying to dynamically configure the topics, services list passed to pyros instance using [setup method.](https://github.com/asmodehn/pyros/blob/master/pyros/pyros_ros.py#L146). But the new instance is not able to [extract topics](https://github.com/asmodehn/pyros/blob/master/pyros/pyros_client.py#L139). Call to services works with this renewed configuration.

Here is my piece of code which initializes a pyros_ros instance:
```
class PyrosClientOnROS(object, ):
    #TODO find out a way to get namespace without rospy preferably
    def __init__(self, enable_cache=False):
        #lets set minimum rosinstnace to get global namespace
        # self.rosInstance = PyrosROS(kwargs={'topics':['/abc'],'services':['/get_global_namespace'],'params':['/def'],'enable_cache': enable_cache})  # careful assuming the topic fullname here
        self.rosInstance = PyrosROS(kwargs={'topics':['/chatter','/.*'],'services':['/get_global_namespace'],'params':['/def'],'enable_cache': enable_cache})  # careful assuming the topic fullname here
        # setting up ros instance
        # TODO: Catch exceptions here
        cmd_conn = self.rosInstance.start()
        self.client = PyrosClient(cmd_conn)
        #wait for the client to come up
        time.sleep(3)
        self.__namespace() #sets up self.ros_ns to namespace 
        #prepend namespace to all topics from flyt
        #TODO how about adding non flyt stuff here
        #TOPICS, SERVICES, PARAMS are lists imported from config.py
        Topics = [self.ros_ns+x for x in TOPICS]
        Services = [self.ros_ns+x for x in SERVICES]
        Params = [self.ros_ns+x for x in PARAMS]

        #Reinitialize rosinterface with list of services and topics prepended with namespace
        self.rosInstance.setup(topics= Topics, services=Services , params= Params, enable_cache= enable_cache)  # careful assuming the topic fullname here
        cmd_conn = self.rosInstance.start()
        self.client = PyrosClient(cmd_conn)
``` 
We had discussed this earlier and I was using setup method until I realized this issue. Now I have moved on to regular expressions as you had suggested earlier like '/.*/servicename' to pass a topics/services with dynamic namespace. 

We also had discussed about passing '/.*/' in topics list so that everything is exposed from ros. But when I tried this with some ~20 ros topics running, the rosinterface got stuck in creating rosinterface topics. @asmodehn have you tested this approach?

asmodehn, 2016-10-23:
I don't remember testing this extensively... There are likely some bugs lying around...
Probably a rosinterface bug. I added this method a bit later, and it looks like there isn't any unit test to confirm it works as expected yet. I don't do anything related to setup [here](https://github.com/asmodehn/pyros/blob/master/pyros/rosinterface/rostests/testRosInterface.py)

Please send a PR if you are able to isolate and reproduce this problem in a meaningful unit test. It would be a useful addition to the tests already there.

Just1045523143, 2020-10-15:
hello @dhirajdhule, in your source code, why
```
#wait for the client to come up
 time.sleep(3)
```

dhirajdhule, 2021-02-16:
@Just1045523143  going by the comments, it was to wait for the client to come up (which would be happening in separate thread).