pyros-dev/pyros #109: Do not use rospy MasterAPI as a reference

asmodehn, 2016-11-04:
Currently in the code the rospy master getSsytemState() datastructure is used as a reference.

However this list of list of list is quite inefficient, and unnecessarily complex to manipulate...
We should establish and use a different  data format, likely based on dictionaries (for optimized access) and sets (for optimisedset operations)

stonier, 2016-11-04:
You are a bit slow on this decision ;).  This ideally should be a useful
functionality provided by the connection cache.

asmodehn, 2016-11-04:
The connection cache (and proxy) already provide data in a different format.
It's the function calls inside pyros that do some unneeded gymnastic between the two data formats...