pyros-dev/pyros #122: user behavior driven testing ?

asmodehn, 2016-11-20:
https://www.youtube.com/watch?v=ZEZBZFi_e88

stonier, 2016-11-20:
Really, cucumber!? What is with these names!

stonier, 2016-11-20:
Python people must hate life...they carry dieting extremes into their programming.

asmodehn, 2016-11-20:
There is an implementation called lettuce as well... well if you want more
women into tech fields, some sacrifices need to be made :'/

stonier, 2016-11-20:
Most women I know actually like chicken, ice cream, chocolate - if anything more than we do because of prolonged periods of enforced abstinence due to a society expecting them to look wonderful! So I suspect they *hate* celery/cucumber/lettuce, but they may *hate more* having to work with ice cream, chocolate and chicken while forbidden to partake of them :P

asmodehn, 2016-11-21:
Since this package is moslty a python developer focused package, BDD doesn't seem suited for it.

closed 2016-11-21 by asmodehn