pyros-dev/pyros #131: How to use this with Flask, Django or Channels

cjds, 2017-03-06:
Is there any tutorial about how to use pyros with any of these system. This is mentioned in your description as part of the functionality, but I can't figure out how to use it.

Cross-posted here : https://github.com/asmodehn/pyros-setup/issues/47

asmodehn, 2017-03-07:
I am using pyros with flask in https://github.com/asmodehn/rostful.
What is your goal, what are you trying to achieve ? Maybe rostful can do what you need to...

cjds, 2017-03-07:
I'm trying to run ros with an existing Django app ideally

asmodehn, 2017-03-27:
I have no experience with django, but I suppose that the way to do that would be similar to flask. Run another process (a ros node, nothing to do with WSGI), and use the client side from the webserver.
Good Luck !

asmodehn, 2019-07-04:
Has there been any progress on this ?

closed 2019-07-18 by cjds