pyros-dev/pyros #132: No module named pyros_utils

cjds, 2017-03-06:
I tried to use this library like so:

    pip install pyros
    python
    >>>> import pyros

I get the following error:

>  WARNING:root:ZMQ : Protobuf message implementation not found. Using pickle based protocol
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/cjds/development/robot_configuration_interface/env/local/lib/python2.7/site-packages/pyros/__init__.py", line 38, in <module>
>     from .pyros_ctx_server import pyros_ctx
>   File "/home/cjds/development/robot_configuration_interface/env/local/lib/python2.7/site-packages/pyros/pyros_ctx_server.py", line 9, in <module>
>     from .pyros_mock import PyrosMock
>   File "/home/cjds/development/robot_configuration_interface/env/local/lib/python2.7/site-packages/pyros/pyros_mock.py", line 3, in <module>
>     from .rosinterface.mock import MockInterface
>   File "/home/cjds/development/robot_configuration_interface/env/local/lib/python2.7/site-packages/pyros/rosinterface/__init__.py", line 41, in <module>
>     from .subscriber_if_pool import RosSubscriberIfPool
>   File "/home/cjds/development/robot_configuration_interface/env/local/lib/python2.7/site-packages/pyros/rosinterface/subscriber_if_pool.py", line 9, in <module>
>     import pyros_utils
> ImportError: No module named pyros_utils
> >>> import pyros_utils
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ImportError: No module named pyros_utils
>

asmodehn, 2017-03-07:
Hello. This is interesting, thanks for reporting the problem.

In my case, in a fresh virtualenv (with `--system-site-packages`), just after `pip install pyros`, this happens : 
```
(pyros0demo) alexv@AlexV-Linux:~$ python
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyros
WARNING:root:ZMQ : Protobuf message implementation not found. Using pickle based protocol
WARNING:pyros.rosinterface:loading pyros_setup and configuring your ROS environment
WARNING:pyros_setup:Dynamic PyROS setup starting...
WARNING:pyros_setup.ros_setup: => Pyros_setup v0.2.0 Emulating ROS setup now for distro indigo and workspaces ()
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo to CMAKE_PREFIX_PATH
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo/share to ROS_PACKAGE_PATH
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo/bin to PATH
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo/lib to LD_LIBRARY_PATH
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo/lib/pkgconfig to PKG_CONFIG_PATH
WARNING:pyros_setup.ros_setup:Prepending path /opt/ros/indigo/lib/python2.7/dist-packages to sys.path
Re-adding /opt/ros/indigo/lib/python2.7/dist-packages in front of sys.path
WARNING:pyros_setup.ros_setup: => ROS setup emulation done.
WARNING:pyros_setup:Dynamic PyROS setup done.
WARNING:rosout:Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
>>> >>> pyros.__version__
'0.3.1'
```

pyros-utils is a ros package, and not available as a pip package, so :

- it needs to be installed in your ROS envrionment
- you ROS environment should be already available (loaded it with pyros_setup)

In my case pyros_utils is available as a package in my ROS install : 
```
>>> import pyros_utils
>>> pyros_utils.__file__
'/opt/ros/indigo/lib/python2.7/dist-packages/pyros_utils/__init__.pyc'
```

and my configuration for pyros_setup is the one automatically configured on first run (since my virtualenv was empty, pyros_setup autodetect my ROS setup and write the configuration : 
```
(pyros0demo) alexv@AlexV-Linux:~/.virtualenvs/pyros0demo/var/pyros_setup-instance$ cat pyros_setup.cfg 

# default configuration generated by pyros-setup
# Usage from python :
# import pyros_setup
# pyros_setup.configurable_import().configure().activate()
#
# Fill in your workspaces here, if you want to dynamically import ROS packages from it.
WORKSPACES=[]

# ROS distribution. Change this value to the ROS distribution you want to use with this environment.
DISTRO='indigo'
```

Please do let me know if you find out what happened differently in your case.
I am still unsure of your environment... it looks like you have a `env/` prefix but not a virtualenv, nor a catkin workspace ? I would be interested to know more about your setup...

cjds, 2017-03-07:
It was a virtualenv. I just named the folder differently. I was trying to do this in pure Python. I had a roscore running, but I've never installed pyros_utils. 

I would recommend adding this to documentation as a dependency. Also, adding some more information to the ROS wiki http://wiki.ros.org/pyros. I had no idea it had to be installed

closed 2017-03-07 by cjds