pyros-dev/pyros #142: Pub/Sub communication example

hanggaga, 2019-02-15:
I'm newbie. I need a simple code to publisher/subscriber information with the topics I want. please help!.

asmodehn, 2019-02-15:
Indeed, currently the [examples](https://github.com/pyros-dev/pyros/tree/master/examples) are quite empty and would benefit from a PR... You may want to give it a try though, in case the instructions there still work with minimal changes.

Otherwise you can get inspiration from rostful code.The backend class does access to topics information and pub sub messages : https://github.com/pyros-dev/rostful/blob/master/rostful/api_0_1/flask_views.py#L105 but you will have to untangle the web/json formatting code from the pyros API call.

But depending on what you want to do, this may or may not be the best approach.
If you re are indeed a newbie, you should first learn from the [ROS tutorials](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29).

Pyros is about interfacing multiprocess systems(like ROS, or a web backend), using python.
Interfacing between python and ROS can overall be done in 5 different ways :
1. interfacing python with ROS with python libraries in one process (simple & lots of support from ROS community), the usual usecase for ROS.
2. interfacing pure python with ROS over request/response (not simple, but doable with a bit of practice), the ROS node being the server.
3. Same as 2, but the pure python is the server part, ROS node the client part. This ended up to be way easier than 2.
4. interfacing pure python with ROS for any kind of communication patterns (definitely harder to pull off, especially if trying to achieve some level of performance & security and simplicity of deployment), the ROS node being the server.
5. Same as 3 but the pure python is the server part, ROS node the client part. Might be easier than 4, but I haven't tried yet.

Pyros is currently doing 2, and I have been working on and off, when time/resources permit, on a few packages for 3, as the code would be much simpler (except a few python tricks) and more maintainable overall.

hanggaga, 2019-02-20:
Thank you for your prompt reply.

closed 2019-02-20 by hanggaga